AntonioMR/Nano-Play-Board #5: Propuesta - Añadir una tira de 4 pines hembra para un sensor de ultrasonidos

josejuansanchez, 2016-06-27:
Estaría bien incluir una tira de 4 pines hembra para poder conectar un sensor de ultrasonidos directamente en la placa.

En la imagen se muestra el tipo de sensor de ultrasonido con los que la placa podría ser compatible.

[image]

josejuansanchez, 2016-07-21:
No tengas en cuenta esta propuesta, creo que es mejor utilizar pines macho como los que usan en esta placa:

[image]

AntonioMR, 2016-07-21:
No me parece mala idea poner un zócalo hembra para insertar directamente el modulo, seria mas practico que tener que andar poniendo cables.
La sigo teniendo en mente.

AntonioMR, 2016-08-07:
Añadido en la versión 0.2

closed 2016-08-07 by AntonioMR