AntonioMR/Nano-Play-Board #6: Propuesta - Añadir una tira de 4 pines hembra para conectar un módulo bluetooth

josejuansanchez, 2016-06-27:
Se podrían incluir 4 pines hembra para conectar directamente un módulo de bluetooth sobre la placa.

La placa podría ser compatible con los módulos bluetooth HC-05 y HC-06.

[image]

josejuansanchez, 2016-07-21:
Creo que esto no sería necesario, he estado haciendo pruebas con un módulo bluetooth HC-05 conectado a los pines  **D7 (Rx)** y **D8 (Tx)** y creo que así está perfecto.

Lo que sí podría ser interesante es añadir dos pines macho adicionales de **+5v** y **GND** junto a los pines **Rx** y **Tx**.

AntonioMR, 2016-07-21:
La idea original de esos pines era conectarse con otras placas iguales para comunicarse. Sin duda con los pines de alimentación mucho mas practico para conectar directamente el modulo blueooth.

closed 2016-07-21 by AntonioMR