AntonioMR/Nano-Play-Board #9: Propuesta - Pinout para la próxima versión

josejuansanchez, 2016-07-21:
Aquí dejo una posible propuesta de cómo podría ser el *pinout* de la próxima versión.
He resaltado en negrita los cambios que he propuesto respecto a la versión actual.

## Pines
Referencia | Pin 
----------|-------
| **Rx**. Para el módulo Bluetooth | **D0** |
| **Tx**. Para el módulo Bluetooth | **D1** |
| Activación de la primera columna de la matriz | D2 |
| Altavoz piezoelectrónico | D3 (PWM) |
| Activación de la segunda columna de la matriz | D4 |
| **Servo**. (Pines macho) | **D5 (PWM)** |
| **Servo**. (Pines macho) | **D6 (PWM)** |
| **Sensor IR**. (Pines macho) | **D7** |
| **Sensor IR**. (Pines macho) | **D8** |
| Pin Rojo RGB | D9 (PWM) |
| Pin Verde RGB | D10 (PWM) |
| Pin Azul RGB | D11 (PWM) |
| Dato de entrada al registro de desplazamiento | D12 |
| Reloj de carga serie del registro de desplazamiento | D13 |
| Fotoresistencia | A0 |
| Potenciometro | A1 |
| Activación de la cuarta columna de la matriz | A2 |
| Activación de la quinta columna de la matriz | A3 |
| **Activación de la tercera columna de la matriz** | **A4** |
| **Reloj de carga de las salidas del registro de desplazamiento** | **A5** |
| **Pulsador** | **A6** |
| **¿Pulsador o Sensor de temperatura?** | **A7** |

AntonioMR, 2016-07-21:
Muchas gracias por todas las sugerencias, sin duda mejoraran mucho la placa y la haran mas util a la hora de enseñar cosas con ella. Te comento lo que tenia en mente para la evolucion de la placa. 
Tenia en mente añadir un segundo registro en serie con el primero para controlar la activación de las columnas de la matriz, de modo que el primer byte enviado es el dato de la columna y el segundo la columna o columnas que se activara. Esto libera 5 pines digitales. de los que tres quería utilizar para el codificador de los pulsadores
Los pines A4 y A5 como son los que implementan el bus I2C tenia pensado aprovecharlos para conectar algún tipo de acelerómetro, sensor Tª o cualquier sensor I2C que se integrara dentro de la placa.
Como el pin D2 soporta una interrupción externa ando dándole vueltas a como utilizarlo para que se pueda aprovechar esta funcionalidad, había pensado en un contador de pulsos o algo así.
Para añadir todo esto en el mismo espacio, todas las resistencias y probablemente los transistores pasen a versiones SMD de un tamaño razonable como para que se puedan soldar con relativa facilidad.

josejuansanchez, 2016-07-21:
La evolución de la placa que propones me parece brillante!

AntonioMR, 2016-08-07:
Finalmente la versión 0.2 de la placa ha quedado con la siguiente distribución de pines

## Pines
Referencia | Pin 
----------|-------
| Tx. Para el módulo Bluetooth | D0 |
| Rx. Para el módulo Bluetooth | D1 |
| Encoder canal A | D2 |
| Altavoz piezoelectrico | D3 (PWM) |
| Reloj de carga de las salidas de los registros | D4 |
| Servo 1. (Pines macho) | D5 (PWM) |
| Servo 2. (Pines macho) | D6 (PWM) |
| Ultrasonidos Echo. (Pines macho) | D7 |
| Ultrasonidos Trig. (Pines macho) | D8 |
| Pin Rojo RGB | D9 (PWM) |
| Pin Verde RGB | D10 (PWM) |
| Pin Azul RGB | D11 (PWM) |
| Dato de entrada al registro de desplazamiento | D12 |
| Reloj de carga serie del registro de desplazamiento | D13 |
| Pulsadores bit 0 | A0 |
| Pulsadores bit 1 | A1 |
| Encoder canal B | A2 |
| Dato sensor Tª/HR  | A3 |
| Linea SDA del bus I2C para conexión con acelerómetro | A4 |
| Linea SCL del bus I2C para conexión con acelerómetro| A5 |
| Fotorresistencia | A6 |
| Potenciómetro | A7 |

josejuansanchez, 2016-09-16:
Creo que no ha sido buena idea usar los pines `D0` y `D1` para el módulo Bluetooth. Tenías razón, nos limita las posibilidades de uso de la placa y no es nada cómodo estar quitando el módulo Bluetooth cada vez que se sube un *sketch*. 

Siento haberte dado la lata para que lo cambiaras :(
Si se puede cambiar en la próxima versión estaría genial.

AntonioMR, 2016-09-16:
En la próxima versión, si todos los pines PWM pasan a controlarse mediante un driver para leds controlado por I2C, se liberarían 6 pines:
2 para puerto serie destinado a comunicaciones inalámbricas (bluetooth, wifi, xbee ...)
1 para la interrupción del acelerómetro
2 para el decodificador de pulsadores y poder controlar hasta 8 pulsadores (un piano? seria genial poder llegar hasta doce y poder tener una octava con los semitonos y todo). Sobre todo pensando en la placa como controladora midi lanzando samplers.
quedaría uno libre que ya se nos ira ocurriendo, quizás para aumentar el numero de pulsadores.

AntonioMR, 2016-09-25:
Una propuesta para la proxima revision:

## Pines
Referencia | Pin 
----------|-------
| Tx. Para la comunicación USB | D0 |
| Rx. Para la comunicación USB | D1 |
| Encoder canal A | D2 |
| Canal de interrupcion del acelerometro | D3 (PWM) |
| Reloj de carga de las salidas de los registros | D4 |
| Tx. Para el módulo Bluetooth | D5 (PWM) |
| Rx. Para el módulo Bluetooth | D6 (PWM) |
| Ultrasonidos Echo. (Pines macho) | D7 |
| Ultrasonidos Trig. (Pines macho) | D8 |
| Pulsadores bit 2 | D9 (PWM) |
| Pulsadores bit 3 | D10 (PWM) |
| Pulsadores bit pulsación | D11 (PWM) |
| Dato de entrada al registro de desplazamiento | D12 |
| Reloj de carga serie del registro de desplazamiento | D13 |
| Pulsadores bit 0 | A0 |
| Pulsadores bit 1 | A1 |
| Encoder canal B | A2 |
| Dato sensor Tª/HR  | A3 |
| Linea SDA del bus I2C para conexión con acelerómetro y driver de LED's | A4 |
| Linea SCL del bus I2C para conexión con acelerómetro y driver de LED's | A5 |
| Fotorresistencia | A6 |
| Potenciómetro | A7 |

Se utilizaría un codificador con prioridad de 16 a 4 para leer el estado de 12 pulsadores distribuidos como teclado de piano formando una octava.

Los 2 servos, el buzzer y el led RGB pasarían a estar controlados por el driver para LED PCA9685 capaz de generar hasta 16 señales PWM independientes. Se utilizarían 5 y el resto se pueden utilizar para encender LED's junto a los pulsadores

MiguelAngelLV, 2016-11-21:
Para una futura revisión, se podría cambiar el LED RGB por uno basado en el ws2812, estos requieren un solo pin PWM. Quizás sea más fácil que usar el bus del i2c o dejarlo libre para otro dispositivo.

También, para ahorrar 1 GPIO en los botoenes, podríamos usar este esquema:

[image]

Con 3 PINs podríamos tener hasta 6 botones. Envías corriente por A0, si recibes en A1 estás pulsando el B0 y si te llega por el A2 estás pulsando el B1.

Ahora cambias el modo y envías por A1, si recibes por A0 estás pulsando B2 y si recibes por A2 estás pulsando B3.

Igualmente podríamos hacerlo enviando por A2 y comprobando A0 y A1 para obtener 2 botones más.

Tendríamos que tener unos delays de microsegundos (pongamos 20 microsegundos por par de botones), pero no creo que sea apreciable.

AntonioMR, 2016-11-21:
Interesantes ambas propuestas, tomo nota.

MiguelAngelLV, 2016-11-21:
Chachi. Bueno, se me había olvidado comentar que los LEDs ws2812 puedes ponerlos en serie y conectar todos los que quieras. Así con un solo PIN puedes poner todos los LEDs que te de la gana. Yo creo que con 4 podrían hacerse cosas muy chulas.

AntonioMR, 2017-01-04:
Propuesta para la V3 de la placa

| Referencia | Pin |
| --- | --- |
| Tx. Para la comunicación USB | D0 |
| Rx. Para la comunicación USB | D1 |
| Encoder canal A | D2 |
| Canal de interrupcion del acelerometro | D3 (PWM) |
| Reloj de carga de las salidas de los registros | D4 |
| Encoder canal B | D5 (PWM) |
| Infrarrojos Izquierdo | D6 (PWM) |
| Ultrasonidos Echo. (Pines macho) | D7 |
| Ultrasonidos Trig. (Pines macho) | D8 |
| Infrarrojos Derecho | D9 (PWM) |
| Tx. Para el módulo Bluetooth  | D10 (PWM) |
| Rx. Para el módulo Bluetooth  | D11 (PWM) |
| Dato de entrada al registro de desplazamiento | D12 |
| Reloj de carga serie del registro de desplazamiento | D13 |
| Pulsadores bit 0 | A0 |
| Pulsadores bit 1 | A1 |
| Pulsadores - Detección pulsación | A2 |
| Dato sensor Tª/HR | A3 |
| Linea SDA del bus I2C para conexión con acelerómetro y driver de LED's | A4 |
| Linea SCL del bus I2C para conexión con acelerómetro y driver de LED's | A5 |
| Fotorresistencia | A6 |
| Potenciómetro | A7 |

Incluiría un driver para Led controlado por I2C para 8 canales (4 servos, Led RGB)